ArminJo/arduino-test-compile #2: Support multiple platform urls

Legion2, 2020-03-23:
Some boards depend on other boards and so require multiple boards to be installed.
Example [CorsairLightingProtocolBoards](https://github.com/Legion2/CorsairLightingProtocolBoards), it's a wrapper for other boards.

This require the installation of multiple boards and multiple platform urls. Also it's not possible to extract all the platforms that must be installed from the fqbn, because it don't provide information about the dependencies of the board.

ArminJo, 2020-03-27:
Please provide a complete example, I think I can then extend the action.
Do you only need additional URLs or also additional boards to be installed manually?

Legion2, 2020-03-27:
This is my current setup with arduino cli, it requires me to install one library and three boards with two additional platform urls.
https://github.com/Legion2/CorsairLightingProtocol/blob/160790313de5a2d8456637be4f0ea744ef0b0a99/.github/workflows/push.yml#L29-L51

I also want to specify the installed cores and libraries version.

ArminJo, 2020-03-27:
The [script version](https://github.com/ArminJo/arduino-test-compile#multiple-boards-with-parameter-using-the-script-directly) should do it for you, if you need additional steps to have more than one board installed.
Otherwise the action is sufficient, since I added a parameter arduino-platform to be able to specify core versions.

Specifying library version and multiple URL's is (and was) straightforward and now documented :-).

ArminJo, 2020-03-28:
Hi Leon,
I modified the script. Now multiple board specs are allowed as [arduino-platform parameter](https://github.com/ArminJo/arduino-test-compile#arduino-platform).
This should solve your request completly.

If **your tests** are also successful, I will release this as a new version.
Thanks
Armin

Legion2, 2020-03-28:
Ok, thanks for your work, will test it later.

Legion2, 2020-03-28:
How can I compile exactly one sketch that I specify, I don't want to explicitly exclude all others.

Legion2, 2020-03-28:
@ArminJo your Action works very well. I'm now looking for a better way the specify all the sketches [example workflow](https://github.com/Legion2/CorsairLightingProtocol/blob/arduino-test-compile/.github/workflows/push.yml)

closed 2020-03-28 by Legion2

ArminJo, 2020-03-29:
Instead of sketch-name: ./examples/${{ matrix.sketch }}/${{ matrix.sketch }}.ino try sketch-name: ${{ matrix.sketch }}.ino. the script does a find for it.
In the new version I changed `required-libraries` from **space** to **comma* separated list 
and you can use a comma separated exclude list :-)