ArminJo/arduino-test-compile #4: Support examples-inlcude

Legion2, 2020-03-29:
I want to explicitly state which sketches should be compiled. Then you can directly see in the workflow definition which examples are compiled. This also reduces duplicate specification, when you have sketches for specific boards which must be excluded from every other boards compilation.

ArminJo, 2020-03-29:
I understand and will have a look...

ArminJo, 2020-03-29:
You do not want to specify full pathname I guess?

Legion2, 2020-03-29:
Would be nice

ArminJo, 2020-03-29:
Please test it.
see https://github.com/ArminJo/arduino-test-compile#multiple-boards-with-parameter

Legion2, 2020-03-29:
@ArminJo I can't find the `examples-inlcude` option. When you force push to master, it's hard to track the changes you made.

ArminJo, 2020-03-29:
You are right but I want avoid an endless commit log!
The option is `sketch-names`

Legion2, 2020-03-30:
Ok, cool can you trim the values after splitting at `,` so it is possible to specify the list using [this syntax](https://github.com/Legion2/CorsairLightingProtocol/blob/arduino-test-compile/.github/workflows/push.yml#L24-L35). This introduces white space for each new line.

ArminJo, 2020-03-30:
Done :-)

Legion2, 2020-03-30:
Cool, thanks for your help.

closed 2020-03-30 by Legion2