ArminJo/arduino-test-compile #5: Readme has wrong examples-build-properties description

Legion2, 2020-03-29:
The description of [examples-build-properties](https://github.com/ArminJo/arduino-test-compile/blob/3e6885559e60747266c85c396584d696cba6da5b/README.md#examples-build-properties) in the readme does not explain format of the configuration and the example makes no sense. Looks like the example was copy-pasted from `examples-exclude`.

Legion2, 2020-03-29:
The EXAMPLES_BUILD_PROPERTIES is printed without the first `{`
https://github.com/ArminJo/arduino-test-compile/blob/3e6885559e60747266c85c396584d696cba6da5b/arduino-test-compile.sh#L190

ArminJo, 2020-03-29:
THANKS,
readme is fixed.

Legion2, 2020-03-29:
also the json is not parsed correctly, use jq instead of regex.

ArminJo, 2020-03-29:
what means "not correctly", it works for all my tests.

> The EXAMPLES_BUILD_PROPERTIES is printed without the first {
is fixed

Legion2, 2020-03-29:
You can not specify a yaml object as argument to the action as it is suggested by the updated readme:
`Your workflow file was invalid: The pipeline is not valid. .github/workflows/push.yml (Line: 40, Col: 13): A mapping was not expected`

Legion2, 2020-03-29:
https://github.com/Legion2/CorsairLightingProtocol/runs/543092637?check_suite_focus=true#step:4:99 here is the output of my workflow and the first `{` is not printed

Legion2, 2020-03-29:
It also happens in your workflow https://github.com/ArminJo/Arduino-RobotCar/runs/542840082?check_suite_focus=true#step:3:112

ArminJo, 2020-03-29:
The trailing { was fixed a few mitutes ago
https://github.com/ArminJo/Github-Actions/runs/543130754?check_suite_focus=true#step:4:93

ArminJo, 2020-03-29:
> You can not specify a yaml object as argument to the action as it is suggested by the updated readme:
> `Your workflow file was invalid: The pipeline is not valid. .github/workflows/push.yml (Line: 40, Col: 13): A mapping was not expected`

Can you be more specific?

Legion2, 2020-03-29:
I tried the examples-build-properties syntax as described in the readme:
```yaml
examples-build-properties:
 WhistleSwitch:
    -DDEBUG
    -DFREQUENCY_RANGE_LOW
  SimpleFrequencyDetector:
    -DINFO
```
https://github.com/Legion2/CorsairLightingProtocol/blob/f6ba148ba2efea0ac8030bb91e6a657601f9a403/.github/workflows/push.yml#L39-L46
and got the error 
https://github.com/Legion2/CorsairLightingProtocol/runs/543117357?check_suite_focus=true

ArminJo, 2020-03-29:
I seems, that directly specifying at with: is not possible.
in the readme there is:
`examples-build-properties: ${{ toJson(matrix.examples-build-properties) }}`

or it is specified in the include section.
```
        include:
...
            examples-build-properties:
              WhistleSwitch:
                -DDEBUG
                -DTRACE
```

I found no way to specify it directly 👎

ArminJo, 2020-03-29:
I found it :-)
Use escaped double quotes.

`examples-build-properties: "{ \"WhistleSwitch\": \"-DDEBUG -DFREQUENCY_RANGE_LOW\", \"SimpleFrequencyDetector\": \"-DINFO\" }"`

Legion2, 2020-03-29:
I used singe quotes and then it worked:
```yaml
examples-build-properties: '{"DebugSketch": "-DDEBUG -DVERBOSE -DPRINT_COMMAND=true -DPRINT_RESPONSE=true -DPRINT_LOOP=true -DPRINT_UPDATE=true"}'
```
https://github.com/Legion2/CorsairLightingProtocol/blob/arduino-test-compile/.github/workflows/push.yml#L39

Legion2, 2020-03-30:
can you update the readme, because the current example syntax does not work

ArminJo, 2020-03-30:
What exactly is wrong?

Legion2, 2020-03-30:
When applying the configuration as shown in the readme directly in the `with:` section of the action it won't work:
[image]
But the readme doesn't tell you that. It should contain information about the json format used to encode this information.

ArminJo, 2020-03-30:
Now I see

ArminJo, 2020-03-30:
Thanks! I hope it is clearer now.

closed 2020-03-30 by Legion2