ArminJo/arduino-test-compile #7: How to include .c,.h files for custom libraries to still keep the code same as on desktop. Or is there any other way to edit in script and add these to search paths

rnbokade, 2020-05-09:


ArminJo, 2020-05-09:
Please explain! 
What do you mean with custom library? 
Do you see the https://github.com/ArminJo/arduino-test-compile#examples-exclude parameter?

rnbokade, 2020-05-09:
yes, I have seen it and it doesn't solve my problem. I have written fec.c and fec.h and few other such header and c files which I include in my sketch with #include<fec.h> #include<abc.h> on my computer... I am able to compile my sketch because I have kept these files in /Documents/Arduino/Libraries/custom/ on my windows PC. I want to take my same code that runs on the desktop and upload it to GitHub while having CI for it. Read somewhere that when compiling sctech if you keep headers and c files in /src subfolder in your folder it compiles similarly. But it is not doing so In this case

ArminJo, 2020-05-09:
For Arduino you must put all your costom libraries into the example/xyz directory,
like ist is done in https://github.com/ArminJo/ServoEasing/tree/master/examples/RobotArmControl

ArminJo, 2020-05-15:
[Here](https://github.com/ArminJo/arduino-test-compile#using-custom-library) it is explained.
Please test and give me Feedback.

closed 2020-05-18 by ArminJo