ArminJo/arduino-test-compile #8: how to add libraries with URL

tobiasfaust, 2020-05-15:
HI,
ho can i add a library via github url? I need to ad my own libraries and not listed libaries.
Example: 
* Wemos Motor Driver 
* https://github.com/tobiasfaust/i2cdetect

ArminJo, 2020-05-15:
Hallo Tobias,
to use other / custom libraries, you must put all the library files into the sketch directory.
Or copy the library directory in a former GitHub-workflow step to the "$HOME/Arduino/libraries" directory e.g. by using the [Download GitHub Release Action](https://github.com/marketplace/actions/download-github-release).

If you have a working example, please tell me, I will then include it In my documentation.
Gruß aus Köln
Armin

ArminJo, 2020-05-15:
[Here](https://github.com/ArminJo/arduino-test-compile#using-custom-library) it is explained.
Please test and give me Feedback.

tobiasfaust, 2020-05-18:
Hi Armin,
great, it works with 4 Custom Libraries :)
Thanks a lot

Gruss aus Magdeburg :)
Tobias

closed 2020-05-18 by tobiasfaust