ArminJo/arduino-test-compile #10: no compiled binary if sketch are in root folder

tobiasfaust, 2020-05-20:
Hi Armin,

i observed, if i have my ino sketch in root folder, no binary are present after sucessful compiling.
If my sketch are in a subfolder, it works fine.

You can look into my test:
https://github.com/tobiasfaust/test/actions/runs/110417721
look at step "run pwd" and you will see in root folder nd in subfolder "test" the same content.
But only in subfolder the binary exists

If you are confused, i can explain it in more detail in german ;)

ArminJo, 2020-05-20:
Hi Tobi, it works as expected, since in the first case I copy the sketch to a temporary folder, where the results are stored.
see output line:
`Creating directory /home/runner/test and copy ./* to it`
After the action, the temporary folder is gone.
For what reason do you need the result, and where do you expect it?

tobiasfaust, 2020-05-20:
I need the binary to upload it to AWS S3. With my own script i generate a proper json file.#
I need this functionality to automatic flash my tons of ESP8266.

ArminJo, 2020-05-20:
Does it makes sense to copy the bin and elf files back to the original directory???
Is there a reason for not having a directory for the ino file?

tobiasfaust, 2020-05-20:
i would be happy if you told me where i will find the final bin file :)
But Yes, It make sense to put the generated *.bin file in the same folder of the sketch (ino file)

I develop no libraries, i develop final sketches for ESP. And all *.cpp, *.h, and *.ino are located in root folder like the Arduino IDE expect :)

for your interesting please take a look at: https://github.com/tobiasfaust/ESP8266_PumpControl
In updater.cpp i check for a new binary and if yes, i will download, install and restart the ESP

ArminJo, 2020-05-20:
> i would be happy if you told me where i will find the final bin file :)

It is in the script output....

`Creating directory /home/runner/test and copy ./* to it`

> And all *.cpp, *.h, and *.ino are located in root folder like the Arduino IDE expect :)

Tell me if I am wrong, but as far as I know Arduino only accepts <name>.ino in folder <name>. Otherwise I could skip the copying above, but this did not work for me 😢 .

tobiasfaust, 2020-05-21:
Please check the output, at root ./ there are the ino but not the binary file which i´m looking for :(
Maybe you see my mistake :(
(Source: https://github.com/tobiasfaust/test/runs/692864566)

Compiling sketches / examples for board esp8266:esp8266:d1_mini 

Creating directory /home/runner/test and copy ./* to it
Compiling test ✓
arduino-cli compile --verbose --warnings all --fqbn esp8266:esp8266:d1_mini --build-properties compiler.cpp.extra_flags="" /home/runner/test
Compiling test ✓
arduino-cli compile --verbose --warnings all --fqbn esp8266:esp8266:d1_mini --build-properties compiler.cpp.extra_flags="" ./test

> pwd
/home/runner/work/test/test

>   ls -ail ./
812308 -rw-r--r-- 1 runner docker   35149 May 20 13:10 LICENSE
812309 -rw-r--r-- 1 runner docker      93 May 20 13:10 README.md
812310 -rw-r--r-- 1 runner docker      24 May 20 13:10 _Release.h
812491 -rw-r--r-- 1 runner docker 6481378 Apr 16 15:16 arduino-cli_latest_Linux_64bit.tar.gz
812875 -rwxr-xr-x 1 runner docker   11715 May 20 13:10 arduino-test-compile.sh
812311 -rw-r--r-- 1 runner docker       0 May 20 13:10 package-lock.json
812312 -rw-r--r-- 1 runner docker     400 May 20 13:10 package.json
823768 drwxr-xr-x 2 runner docker    4096 May 20 13:11 test
812313 -rw-r--r-- 1 runner docker     554 May 20 13:10 test.ino

>   ls -ail ./test
823768 drwxr-xr-x 2 runner docker    4096 May 20 13:11 .
823713 drwxr-xr-x 5 runner docker    4096 May 20 13:11 ..
812314 -rw-r--r-- 1 runner docker   35149 May 20 13:10 LICENSE
812315 -rw-r--r-- 1 runner docker      93 May 20 13:10 README.md
812316 -rw-r--r-- 1 runner docker      24 May 20 13:10 _Release.h
812317 -rw-r--r-- 1 runner docker       0 May 20 13:10 package-lock.json
812318 -rw-r--r-- 1 runner docker     400 May 20 13:10 package.json
817164 -rw-r--r-- 1 runner docker  260720 May 20 13:11 test.esp8266.esp8266.d1_mini.bin
817165 -rwxr-xr-x 1 runner docker 1224516 May 20 13:11 test.esp8266.esp8266.d1_mini.elf
812319 -rw-r--r-- 1 runner docker     554 May 20 13:10 test.ino

>   ls -ail ../
823712 drwxr-xr-x 3 runner docker 4096 May 20 13:10 .
823697 drwxr-xr-x 6 runner root   4096 May 20 13:10 ..
823713 drwxr-xr-x 5 runner docker 4096 May 20 13:11 test

ArminJo, 2020-05-21:
Hi I stumble over  /home/runner/test and /home/runner/**work**/test/test

tobiasfaust, 2020-05-22:
works :) thanks
Binary file found at  /home/runner/test

closed 2020-05-22 by tobiasfaust