ArminJo/arduino-test-compile #9: Use version branches instead of updating releases

Legion2, 2020-05-20:
You can create a branch named `v2` which will point to the latest 2.x release, this way you don't have to update old releases. This works as long as there are no breaking changes.

ArminJo, 2020-05-20:
Thanks for the very interesting feedback, but I was not aware, that v2 was a valid version number. And In this case I have to update the release on every change? which I do not prefer, if I can avoid it.
But maybe I did not get the idea.

Legion2, 2020-05-20:
v2 is a version wildcard that refers to the latest stable v2.x.x version. I'm not sure if it can be realized with branches but tags should work. Using `v2` instead of `v2.1.1` will ensure that (if you always update v2 to new commit) always the latest `v2` compatible version is used by the actions and you don't have to update all your actions when you release a backwards compatible version.

Legion2, 2020-05-20:
here is an example of another action https://github.com/peaceiris/actions-gh-pages/tree/v3 using this pattern

ArminJo, 2020-05-20:
I want to have it!
But I still do not understand how they manage it.
I see the error `An action could not be found at the URI 'https://api.github.com/repos/ArminJo/arduino-test-compile/tarball/v2'`
I think I have to do a bit of research...

ArminJo, 2020-05-20:
Thanks a lot for the suggestion!
I managed it by accident for the current release, but I am not shure how to achieve this for the next release.
Lets see....

closed 2020-05-20 by ArminJo

Legion2, 2020-05-20:
I think they only created tag and don't releases for the major version number `v2`, `v3`

ArminJo, 2020-05-20:
But creating tags is not straightforward with github and I do not know how to get the src.zip and src.tar.gz files to a tag.
Maybe I can delete the release from the tag leaving the tag alone with the 2 assets. 
I will try it for the next version.
Thanks

Legion2, 2020-05-20:
`git checkout v2.2.1`
`git tag v2`
`git push --tags`
You don't need to add the src.zip, they are generated by GitHub.