ArminJo/arduino-test-compile #12: Why use docker?

matthijskooijman, 2020-07-03:
I just started using this action, and it looks well-polished and complete. However, I wonder, why is docker used?

It seems that that adds quite a bit of overhead. I can see a small gain in isolation, but the entire thing already runs in a virtual machine and really only arduino-cli is installed which is just unpacking a zip, so that has little potential to conflict with other steps in the main VM?

The README already mentions calling the script directly as a faster alternative, but that's not so convenient (and more error-prone, typos in env var names are not flagged I expect). Why not just run outside of docker always (or maybe add a `user_docker: yes/no` configuration?

ArminJo, 2020-07-03:
Github gives you a choice between using Docker or JavaScript. But JavaScript is more complicated and not an alternative for me.

matthijskooijman, 2020-07-03:
Oh wow, I thought that predefined actions would just have the same options as the main workflow file, but looking at the docs, I see now that they indeed need either Docker or Javascript. That's unexpected.

However, I can imagine that you can still write a minimal Javascript action that just gets the inputs and then runs the main shell script to do the actual work. Then the javascript component is minimal and not very complicated. How would that sound?

ArminJo, 2020-07-03:
Sounds good, but my last Javascript was 10 years ago. Do you have an example, which I can modify?

matthijskooijman, 2020-07-03:
I haven't done anything with javascript actions so far, but reading the documentation just now I found this example:

https://docs.github.com/en/actions/creating-actions/creating-a-javascript-action

The main index.js there at least shows how to retrieve inputs.

I also found https://github.com/actions/toolkit/tree/master/packages/exec which shows how to run other commands.

Finally, here's an example on how to structure the javascript when you need
asynchronous actions with await like exec:
https://github.com/actions/javascript-action#change-the-code

Combining those, I would think you'd need something like this:

```
const core = require('@actions/core');
const exec = require('@actions/exec');

async function run() {
  try {
    const args = [
      core.getInput('cli-version'),
      core.getInput('sketch-names'),
      core.getInput('sketch-names-find-start'),
      core.getInput('arduino-board-fqbn'),
      core.getInput('arduino-platform'),
      core.getInput('platform-default-url'),
      core.getInput('platform-url'),
      core.getInput('required-libraries'),
      core.getInput('examples-exclude'),
      core.getInput('examples-build-properties'),
      core.getInput('debug-compile'),
      core.getInput('debug-install')
    ];
    await exec.exec('arduino-test-compile.sh', args);
  } catch (error) {
    core.setFailed(error.message);
  }
}
```

I have not tested this at all, so it will likely not work right away, but might
be a good base. One thing I'm completely unsure of is how the filesystem
environment looks like when this action is run. I *hope* that the entire
arduino-test-compile repository is checked out and set to be the current
directory, then the exec above should work. Then I hope that
`$GITHUB_WORKSPACE` is set, just like with the docker action. Looking at the
script, this might need a `cd $GITHUB_WORKSPACE` somewhere in the script to
make the `find` work as expected, but I'm not entirely sure.

ArminJo, 2020-07-03:
Thanks for the code, I will definitely give it a try, but I may takes a few days, since I am currently very busy.

matthijskooijman, 2020-07-04:
I'm currently trying the above, I'll try to submit a PR soon.

oclyke, 2020-07-14:
Excited to see this solved with #16 
Thanks both of you for these tools 🔥

per1234, 2020-11-03:
It looks like you are already on track to use JavaScript, but, just in case it's of any use, I'll mention that GitHub added a new action type in August: "composite run steps". This allows you to write actions that run shell commands directly in the runner:
https://docs.github.com/en/free-pro-team@latest/actions/creating-actions/creating-a-composite-run-steps-action

I converted one of Arduino's actions from a Docker container to a composite run steps action last week, as well as writing a few composite run steps actions from scratch, and so far haven't had any problems.

ArminJo, 2020-11-04:
Thanks to per1234 🥇, it is now convereted to a "composite run steps" action.

closed 2020-11-04 by ArminJo

matthijskooijman, 2020-11-06:
Looks good, thanks for the tip @per1234 and thanks for the implementation @ArminJo :-)

Just reverted my library back to the standard version (was using a customized version with my PRs applied until now), which works like a charm :-)