ArminJo/arduino-test-compile #17: Cache downloaded arduino-cli and downloaded/installed cores?

matthijskooijman, 2020-07-05:
Currently, this action downloads arduino-cli and installs cores from scratch on every job. When using a matrix, with a few different boards and maybe other build options, or when building a lot of revisions daily, this makes things slower than necessary.

It would be interesting to explore options for caching, in particular the downloaded arduino-cli and installed cores would be candidates for caching.

I'm not sure if this should be handled by the action itself (if so, probably should be possible to disable using a config option, since cache size is limited by GH), or just be configured in the workflow file that calls this action (in which case it would be good to at least include an example in the README).

The default way to do caching is https://github.com/actions/cache, which allows caching a particular directory (which could be the arduino-cli unzip directory and `~/.arduino15` where I think arduino-cli installs cores).

The documentation is slightly vague about how the caching works, but I think that:
 1. Each job generates a cache key.
 2. When a cache exists with that key, it is used.
 3. Otherwise, on older cache (as specified by `restore-keys`) is considered as a basis to generate a new cache content.
 4. Otherwise, no cache is restored (a clean slate for generating new cache content).

AFAIU (but this is where the docs seem vague) only case 3. and 4. create a new cache entry (using the key from 1.), in case 2 the existing cache content is left untouched. IOW, content for a given key is generated once, never updated.

The idea is, I think, that you generate a cache key that describes the contents and then the cache content gets generated once and used as long as that same key is generated. There does not seem to be a way to let the cache content expire, which is a bit tricky (since in this case, the cache content actually depends on external content, e.g. the latest version of arduino-cli and cores will changing which needs a cache update). Maybe one can generate a cache key that includes some info (such as the list of cores to be installed) and also a date (or week number). Then a new cache content will be generated each day. Using `restore-keys`, the content from a previous day can be used as a base, so that generating a new cache content will often be mostly a no-op (when nothing actually turns out to be changed).

Alternatively, maybe one could use the [cache node package](https://github.com/actions/toolkit/tree/master/packages/cache) to manually manage the cache, but that's maybe a bit more involved.

One completely different approach would be to not cache, but have a separate job that just installs arduino-cli and installs cores, that is then used by all other jobs (I think GH supports dependencies between jobs and sharing files as well, or maybe the initial job can create a cache entry with a unique key that is then used by the other jobs). This would not cache between multiple pushes/pullrequests, but at least speed up a single run with a lot of different jobs.

I thought a bit about this, but it seems a bit more involved than I had hoped. I do not really need this right now, so I won't be looking into this any further, but I wanted to at least share my thoughts and initial research.

ArminJo, 2020-07-05:
IIRC the cache was also loaded via network and was only sligtly faster than downloading arduino-cli.

I tried caching when starting with the action, but it did not work as expected. The problem was, that with https://github.com/actions/cache cache was global for the action, but every job of the matrix changed the .arduino15 directory differently.

Currently I am a bit busy, so processing of your valuable Input may take a few days.

matthijskooijman, 2020-07-05:
> IIRC the cache was also loaded via network and was only sligtly faster than downloading arduino-cli.

Ah, that seems likely, though I can imagine that downloading and unpacking the cores and all the tools might actually take longer than using a pregenerated cache.

> I tried caching when starting with the action, but it did not work as expected. The problem was, that with https://github.com/actions/cache cache was global for the action, but every job of the matrix changed the .arduino15 directory differently.

Yeah, so I guess you'd need to either generate the cache once with all cores installed, or generate different caches for each core.

> Currently I am a bit busy, so processing of your valuable Input may take a few days.

I'm just running off my own fork for now, so there's no hurry from my side :-)