ArminJo/arduino-test-compile #18: many boards x many defines

i-make-robots, 2020-07-14:
Hi!  Your Action looks great!

I've got firmware that runs on ... 6 different boards (defined in code and set board).  It also has up to 7 different modes (defined in code) for each of those boards.  I want to have a matrix of every valid combination so that my edits are at least compile-safe for every combination.  Your script looks really close to what I'm after.  

Can you confirm, please?

ArminJo, 2020-07-14:
Hi,
I added [documentation](https://github.com/ArminJo/arduino-test-compile#one-board-with-2x2-compile-parameter-matrix) for this case.

i-make-robots, 2020-07-14:
My goal is to test my sketch against several boards and configurations.   Is there a wild card to exclude all examples?

i-make-robots, 2020-07-14:
https://github.com/ArminJo/arduino-test-compile#single-program-using-sketch-names-parameter still mentions examples, so it has me uncertain.

ArminJo, 2020-07-14:
The parameter name examples-build-properties is historic just use [sketch-names](https://github.com/ArminJo/arduino-test-compile#sketch-names)

i-make-robots, 2020-07-14:
I wish I was grokking this...
What if the define is like "#define A B"?  -DA=B?  -DA B?

ArminJo, 2020-07-14:
I never defined macros at the compiler commandline.

i-make-robots, 2020-07-14:
fwiw, the repo in question is https://github.com/MarginallyClever/Makelangelo-firmware
in configure.h there are several defines for BOARD_TYPE and MACHINE_STYLE.  I'm trying to change these at compile time and the board type.

ArminJo, 2020-08-03:
I guess the issue is solved 😀

closed 2020-08-03 by ArminJo