ArminJo/arduino-test-compile #19: Using custom libraries

arashnrim, 2020-10-26:
Hello!

First off, thanks so much for making this; this can help a lot in making sure source code works as it should before execution! I'm facing a difficulty when using custom libraries; I'm not entirely sure I understand the documentation for implementing custom libraries entirely, so I hope it's alright for me to ask this here.

The following is my workflow file:

```
name: Verify Arduino files
on: [ push ]
  
jobs:
  build:
    name: Verifying compilation of Arduino files
    runs-on: ubuntu-latest
    
    steps:
    - name: Checkout project
      uses: actions/checkout@v2
    
    - name: Checkout XBee library
      uses: actions/checkout@v2
      with:
        repository: andrewrapp/xbee-arduino
        ref: master
        path: CustomLibrary
    
    - name: Compile Arduino files
      uses: ArminJo/arduino-test-compile@v2
      with:
        required-libraries: CustomLibrary
        debug-compile: true
        debug-install: true
```

When getting to installing the custom library, though, the Action fails; here's the important bit:

```
... (everything before finished successfully)
**Install required libraries**
Install libraries CustomLibrary
Error resolving dependencies for CustomLibrary: looking for library: library CustomLibrary not found
Error: Installation of CustomLibrary failed
```

I'd expected the library to load in fine; I presume that something must've gone wrong in my configuration somewhere?

Thanks a lot for reading this! A reply is greatly appreciated.

ArminJo, 2020-10-26:
Which project has the error?

arashnrim, 2020-10-26:
Apologies for the ambiguity! I'm not entirely clear what you're referring to. But if you're talking about which custom library I was trying to use, I was trying to checkout the XBee Library (see the Action above). The `path` I set it as is `CustomLibrary`, which should fit the requirements of `*Custom*`.

If you're referring to a repository, though, it's currently private. Should I make it public for a while?

ArminJo, 2020-10-26:
If you want support, then you should make it public.

ArminJo, 2020-10-26:
Ad set debug-install to true

arashnrim, 2020-10-26:
Right, of course. Here's [my repo](https://github.com/arashnrim/temp) with the Actions workflow in it.

ArminJo, 2020-10-26:
Everything works well 😀 
See my fork

closed 2020-10-26 by ArminJo

arashnrim, 2020-10-26:
Ah, awesome. Thank you for your help! 🙌

ArminJo, 2020-10-26:
You do not need to put the custom libraries you loaded manually in the `required-libraries` list, since they are already loaded now!