ArminJo/arduino-test-compile #22: build-properties are only usable for compiler flags, not for the general case

jackjansen, 2020-12-29:
I'm trying to pass `build.partition=no_ota` to arduino-cli (so that my sketch compiles and links correctly: it uses HTTPS and BLE, so the resultant binary is too big to fit into the available Flash otherwise).

But the command line generated by arduino-test-compile is

```
arduino-cli compile --verbose --warnings all --fqbn esp32:esp32:lolin32  --build-properties compiler.cpp.extra_flags="build.partition=no_ota" --build-properties compiler.c.extra_flags="build.partition=no_ota" --build-properties compiler.S.extra_flags="build.partition=no_ota" /home/runner/iotsaDMXSensor
```

So this works fine (and is actually quite convenient) for compiler flags, and solely for such parameters it is used in the documentation indeed.

Is there a way to pass general `--build-properties` flags?

ArminJo, 2020-12-29:
can arduino-cli recogniize the parameter build.partition=no_ota
What is the commandline you want to have generated

jackjansen, 2020-12-29:
The `--build-properties` accepts any `dotted.name=value` and simply overrides those settings. So by passing, say, `compiler.cpp.extra_flags=-blabla` you will pass the `-blabla` option to cpp. But by setting `build.partition=no_ota` you will set that value so it's going to be passed to (I think) the linker driver or the elf-to-binary-converter or whatever.

For my use case I could revert to not using arduino-test-compile but in stead use manual steps with `arduino-cli` invocations, but that would be a bit of a shame, I really like the convenience of arduino-test-compile for many cases.

If this turns out to be difficult my suggestion is that you at least update the documentation to state that you can basically only pass `-D` and `-U` options through build-properties...

ArminJo, 2020-12-29:
I fear I did not get it.
What is the commandline you want to have generated?

jackjansen, 2020-12-29:
Sorry, I wasn't clear. For this specific use case, I would need

```
arduino-cli compile --verbose --warnings all --fqbn esp32:esp32:lolin32  --build-properties "build.partition=no_ota"  /home/runner/iotsaDMXSensor
```

jackjansen, 2020-12-29:
Actually, now that I think of it there are more arduino-cli arguments that may be useful in specific circumstances. Maybe if you add a parameter `extra-arduino-cli-args` that you just pass verbatim to the command line that would be useful.

ArminJo, 2020-12-29:
> Maybe if you add a parameter `extra-arduino-cli-args` that you just pass verbatim to the command line that would be useful.

Good idea, but what is about the double quotes? are they required around "build.partition=no_ota"

per1234, 2020-12-29:
There is some information here that might be of interest:
https://github.com/arduino/arduino-cli/pull/1044
- Deprecation of the `--build-properties` flag in favor of the new `--build-property` flag
- Handling of quotes.

ArminJo, 2020-12-29:
>     * Deprecation of the `--build-properties` flag in favor of the new `--build-property` flag
Already done, thanks!!!
>     * Handling of quotes.
Thanks again, I was not aware of 👍

ArminJo, 2020-12-29:
@jackjansen 
Please check the latest version. Your `extra-arduino-cli-args` is implemented 😀 
Maybe you must test quotes or escaping quotes with \" to get it working.
Please give me feedback. I would like to to add an meaningful and working example for this parameter.

jackjansen, 2020-12-30:
It works, thanks! 

A meaningful example would probably be specifically this use case: setting the partition table for esp8266/esp32 builds.

Here's a quick cut-and-paste from my iotsaDMXSensor repo (https://github.com/cwi-dis/iotsaDMXSensor/commits/0d228e4e6c53e89317436bd4e64d1abf97352d5b/.github/workflows/build-arduino.yml):

```
    strategy:
      matrix:
        arduino-board-fqbn:
        - esp32:esp32:lolin32
        include:
          - arduino-boards-fqbn: esp32:esp32:lolin32
            extra-arduino-cli-args: "--build-property build.partitions=no_ota --build-property upload.maximum_size=2097152"
    ...
    steps:
    - name: Arduino build
      uses: ArminJo/arduino-test-compile@2e04e8c
      with:
        ...
        arduino-board-fqbn: ${{ matrix.arduino-board-fqbn }}
        extra-arduino-cli-args: ${{ matrix.extra-arduino-cli-args }}

```

This will tell `arduino-cli` to do the `lolin32` build for what the Arduino IDE calls _Tools_ -> _Partition Scheme_ -> _No OTA (Large APP)_.

ArminJo, 2020-12-30:
Thanks, I included it in the Readme 👍

closed 2020-12-30 by ArminJo

per1234, 2021-02-09:
>A meaningful example would probably be specifically this use case: setting the partition table for esp8266/esp32 builds.

@jackjansen this is a fragile and unnecessarily complex approach. The right way to do it is by setting the `PartitionScheme` board option via the FQBN:
```
arduino-boards-fqbn: esp32:esp32:lolin32:PartitionScheme=no_ota
```
See this documentation for details:
https://arduino.github.io/arduino-cli/latest/platform-specification/#custom-board-options

jackjansen, 2021-02-10:
@per1234 that is an interesting option... I wasn't aware of the ability to add  `MENU_ID=OPTION_ID` specifications to the FQBN. Thanks!

Again something that could be useful to add to the Readme, @ArminJo ...

commit 1c1c450c (2020-12-30): Added option extra-arduino-cli-args - closes #22