ArminJo/arduino-test-compile #23: Option to disable "--warnings all"

tomjschwanke, 2021-01-30:
Hi there, I use this action to automatically test my projects on GitHub.

However, I'm running into a problem where I get more warnings and warnings treated as errors on GH actions, that do not appear on my local installation, causing the check to fail even though it compiles fine on my local installation.

Output from local Arduino Pro IDE: 

`-IC:\\Users\\tomsc\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fb_gfx" -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=0 `

Output from actions check:

`-I/home/runner/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fb_gfx -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=maybe-uninitialized -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=0 `

As you can see, the latter one has the extra parameters `-Wall -Werror=all -Wextra`

I noticed the actions check uses `arduino-cli compile --verbose --warnings all --fqbn esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none  /home/runner/WiFiWeatherDisplayEsp32`

Is there a way to turn the `--warnings all` off?
The documentation only lets me append more paramters to the arduino-cli call

ArminJo, 2021-01-30:
Warnings do not terminate compile and are not causing the check to fail.
So turning --warnings all will not help you.

tomjschwanke, 2021-01-30:
The log tells me 

> cc1plus: some warnings being treated as errors

So I was looking on how to disable them

ArminJo, 2021-01-30:
Which log?

tomjschwanke, 2021-01-30:
The check output

tomjschwanke, 2021-01-30:
I invoked arduino-cli manually, once with `--warnings all` and once without it. When I included it, the compilation failed. When I left it out, it compiled fine.

ArminJo, 2021-01-30:
I get no error so you must do something wrong.

tomjschwanke, 2021-01-30:
I'm sorry, forgot to mention the branch. It only happens on my [github-ota](https://github.com/tomjschwanke/WiFiWeatherDisplayEsp32/tree/github-ota)-branch. On the others it gives me warnings but compiles fine. I should probably have a look at the code, but the option would still be nice. Thank you.

closed 2021-01-30 by tomjschwanke