ArminJo/arduino-test-compile #24: Ability to set --warnings to default

mikevanis, 2021-02-09:
First of all, thanks for this excellent action!

I'm trying to set up a workflow to test a library for the ESP32 and the compiler is treating some warnings as errors. I noticed that the `--warnings` flag is set to `all` on arduino-cli. Is there a way to override that and set it back to the default, which is `none`?

ArminJo, 2021-02-09:
You are the second to ask this question. The first one discovered, that the warnings trated as errors were severe!

davidchatting, 2021-02-09:
That's understandable for checking contributed code, but where the warning exist in 3rd party libraries it rather limits the usefulness of this action. That's the problem I have right now, a trival warning in a 3rd party library that breaks my build > https://github.com/davidchatting/Approximate/runs/1862770429?check_suite_focus=true - I have raised a PR with the maintainer, but I have to wait to see if they will do the merge.

ArminJo, 2021-02-09:
And you think with --warnings flag not set to all, this fatal error /home/runner/Arduino/libraries/ListLib/src/ListLib.h:14:21: fatal error: arduino.h: No such file or directory will vanish???

per1234, 2021-02-09:
From https://github.com/ArminJo/arduino-test-compile#extra-arduino-cli-args
>`extra-arduino-cli-args`
This string is passed verbatim without additional quoting to the arduino-cli compile commandline as last argument before the filename. See https://arduino.github.io/arduino-cli/commands/arduino-cli_compile/ for compile parameters.

When there are multiple `--warnings` flags in an `arduino-cli` command line, the last one will override the previous ones. So you should be able to use this input to override the default warning setting if you really need it.

I think it's a good thing to have the default warnings setting be "all" because it promotes writing warning free code. That is always important, but especially so when writing code for ESP32. Even if the developer doen't enable warnings, some of the users of their code will, and will have a bad time if there are warnings. But I do think there are times (though perhaps not this one) when there will be warnings from external code you don't have control over. For this reason, it's good to have a way to control the warning level, but it's also good for it to be a little bit inconvenient for the action user to do so. That's the approach I've taken with Arduino's sketch compilation action as well and I've been happy with the results so far.

ArminJo, 2021-02-09:
@ALL
I have never seen an example of a warning treated as error and accepted if  ---warnings all were disabled.
Please provide an example!

per1234, 2021-02-09:
@ArminJo I'm not sure I understand what you're asking so forgive me if I tell you something you already know.

The ESP32 platform is configured so that the `-Werror=all` flag is used when the Arduino compiler warning setting (`--warnings` in Arduino CLI or **File > Preferences > Compiler Warnings** in Arduino IDE) is set to "more" or "all":
https://github.com/espressif/arduino-esp32/blob/1.0.4/platform.txt#L17-L21
```
compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall -Werror=all
compiler.warning_flags.all=-Wall -Werror=all -Wextra
```

So if you compile this sketch with `arduino-cli compile --warnings default` then it will pass:
```c++
void setup() {}
void loop() {}
int foo() {}
```
but if you compile it with the warnings setting on "more" or "all" then it will fail:
```
C:\Users\per\AppData\Local\Temp\arduino_modified_sketch_412384\sketch_feb09a.ino: In function 'int foo()':
sketch_feb09a:3:12: error: no return statement in function returning non-void [-Werror=return-type]
 int foo() {}
            ^
cc1plus.exe: some warnings being treated as errors
exit status 1
no return statement in function returning non-void [-Werror=return-type]
```

If you're asking for a real life example of this happening, I can probably dig one up because this popped up recently in the Arduino IoT Cloud support libraries when ESP32 support was added, but the warnings were easily fixed in the library code so we were able to leave the warning setting at "all"..

per1234, 2021-02-09:
I did a little looking at the IoT Cloud stuff for the real life example and didn't find it, but I did find one here:
https://github.com/107-systems/107-Arduino-UAVCAN/runs/1760023988?check_suite_focus=true#step:4:26

mikevanis, 2021-02-09:
@per1234, that's exactly what I've realised. `-Werror=all` is present in ESP32 when arduino-cli's flag `--warnings all` is set. Many third-party libraries have warnings that fail on this action, but compile albeit with warnings when using arduino-cli or the IDE.

I think there's a separate discussion here as to whether warnings should always trigger fails in test suites. I believe that flexibility is useful sometimes. It's fair enough that `--warnings all` is the default setting, as the purpose of this action is to run strict tests. But the world of ESP32 isn't as clean as we'd like it to be. So onwards!

On my action, I tried @per1234 's suggestion and indeed specifying `--warnings default` as an `extra-arduino-cli-args` overrides the first `--warnings all`. I'll go ahead and close this since I can indeed set --warnings to default 💯  

FYI for anyone else looking,`extra-arduino-cli-args` is currently available only on master, but hopefully it's included in the next release 👍

closed 2021-02-09 by mikevanis

per1234, 2021-02-09:
>Many third-party libraries have warnings that fail on this action, but compile albeit with warnings when using arduino-cli or the IDE.

That depends on what the user has their warning level set to. I always set my warnings to "all" in both Arduino IDE and Arduino CLI because I find them to be very useful for catching problems in my own code. It's annoying when other people's libraries have warnings warnings because then I have to either fix them in the library code (which is cool, but sometimes too much of a distraction when I just want to focus on my project) or else lower my warnings setting just so I can compile for an ESP32 board.

mikevanis, 2021-02-09:
Yes, that makes sense. Most people just download the ESP32 core with the boards manager and have it set to default, so that's probably why many libraries don't bother fixing warnings...

ArminJo, 2021-02-09:
Thanks for the discussion, I added it to the documentation now https://github.com/ArminJo/arduino-test-compile#extra-arduino-cli-args.