ArminJo/arduino-test-compile #27: Unexpected input(s) 'extra-arduino-cli-args'

DustinWatts, 2022-05-30:
When try to set `"--warnings default"` as per the documentation: https://github.com/ArminJo/arduino-test-compile#extra-arduino-cli-args

I get the follwing error when I try to run the workflow:

```
Unexpected input(s) 'extra-arduino-cli-args', valid inputs are ['cli-version', 'sketch-names', 
'sketch-names-find-start', 'arduino-board-fqbn', 'arduino-platform', 'platform-default-url', 
'platform-url', 'required-libraries', 'sketches-exclude', 'build-properties',
'set-build-path', 'debug-compile', 'debug-install']
```

This fails the workflow run. Becasue some of the liraries used give warnigs.

This is the part of my .yaml that includes the `extra-arduino-cli-args`:

```
- name: Test compile for Arduino
   uses: ArminJo/arduino-test-compile@v3.0.0
   with: 
        arduino-board-fqbn: esp32:esp32:esp32:PartitionScheme=no_ota
        extra-arduino-cli-args: "--warnings default"
        platform-url: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
        required-libraries: ArduinoJson
```

per1234, 2022-05-30:
The problem is you are using an outdated version of the action. This [input was added](https://github.com/ArminJo/arduino-test-compile/commit/1c1c450ce5d36690cf6e268e70b90e045ab0c087) in [the `v3.2.0` release](https://github.com/ArminJo/arduino-test-compile/releases/tag/v3.2.0) of the action, but you are using `v3.0.0`.

So you should change this line of your workflow:

```yaml
   uses: ArminJo/arduino-test-compile@v3.0.0
```

To either of:

```yaml
   uses: ArminJo/arduino-test-compile@v3.2.0
```

(for the greatest stability of the workflow)

or:

```yaml
   uses: ArminJo/arduino-test-compile@v3
```

(if you want to always use the latest available version within the 3.x version series)

DustinWatts, 2022-05-30:
Thanks @per1234 ! That solved the issue of the error I was getting!

closed 2022-05-30 by DustinWatts