ArminJo/arduino-test-compile #29: Specify -zmuldefs for compiler options

justcallmekoko, 2022-08-20:
How would someone add `-zmuldefs` for `compiler.c.elf.libs.esp32s2` like what's shown in platform.txt

ArminJo, 2022-08-20:
Can't you use `extra-arduino-cli-args`?

justcallmekoko, 2022-08-21:
@ArminJo I did try that but I don't know if I got the syntax correct because I kept getting "xyzfunction first defined here" errors which I overcome with `-zmuldefs`

justcallmekoko, 2022-08-21:
Just for some more context, I've accomplished what it is I am trying to do with your action but with bash and it looks pretty gross.
```
    - name: Modify platform.txt
        run: |
          echo "Chicken"
          for i in $(find /home/runner/.arduino15/packages/esp32/hardware/esp32/ -name "platform.txt"); do
            sed -i 's/compiler.c.elf.libs.esp32s2=/compiler.c.elf.libs.esp32s2=-zmuldefs /' "$i"
            sed -i 's/compiler.c.elf.libs.esp32=/compiler.c.elf.libs.esp32=-zmuldefs /' "$i"
            cat "$i" | grep compiler.c.elf.libs.esp32s2
            cat "$i" | grep compiler.c.elf.libs.esp32
          done
```