ArminJo/arduino-test-compile #31: Weird result with sketch who has dot in name

LioRei, 2022-08-21:
Hi,

I've tried to compil my sketch but the script doesn't detect the name of the sketch/directory correctly.

I think the bug is between lines 342 and 346. I've not been able to reproduce on my computer.

Output of the job:
```
Compiling sketches / examples for board esp32:esp32:esp32:FlashFreq=80

declare -a SKETCH_NAMES_ARRAY=([0]="Arduino.SignalR.Client.Example.ino")
declare -a SKETCHES=([0]="./Arduino.SignalR.Client.Example/Arduino.SignalR.Client.Example.ino")

Creating directory /home/runner/Arduino and copy ./Arduino.SignalR.Client.Example/* to it
Compiling Arduino ✖
arduino-cli compile --verbose --warnings all --fqbn esp32:esp32:esp32:FlashFreq=80   /home/runner/Arduino
Error: Compile of Arduino  failed
Error opening sketch: main file missing from sketch: /home/runner/Arduino/Arduino.ino 

Compile of Arduino  failed
```

On my computer:
```
root@DESKTOP-9TK8FUL:/home/lionel# SKETCH_PATH=$(dirname "/a/b/Arduino.SignalR.Client.Example/Arduino.SignalR.Client.Example.ino")
root@DESKTOP-9TK8FUL:/home/lionel# SKETCH_DIR=${SKETCH_PATH##*/}
root@DESKTOP-9TK8FUL:/home/lionel# SKETCH_FILENAME=$(basename "/a/b/Arduino.SignalR.Client.Example/Arduino.SignalR.Client.Example.ino")
root@DESKTOP-9TK8FUL:/home/lionel# SKETCH_EXTENSION=${SKETCH_FILENAME##*.}
root@DESKTOP-9TK8FUL:/home/lionel# SKETCH_BASENAME=${SKETCH_FILENAME%%.$SKETCH_EXTENSION}
root@DESKTOP-9TK8FUL:/home/lionel# echo $SKETCH_PATH
/a/b/Arduino.SignalR.Client.Example
root@DESKTOP-9TK8FUL:/home/lionel# echo $SKETCH_DIR
Arduino.SignalR.Client.Example
root@DESKTOP-9TK8FUL:/home/lionel# echo $SKETCH_FILENAME
Arduino.SignalR.Client.Example.ino
root@DESKTOP-9TK8FUL:/home/lionel# echo $SKETCH_EXTENSION
ino
root@DESKTOP-9TK8FUL:/home/lionel# echo $SKETCH_BASENAME
Arduino.SignalR.Client.Example
```

I don't know why on the build machine the output of SKETCH_BASENAME is "Arduino" only.

ArminJo, 2022-08-21:
Please provide a link to the output of your action and specify          
` debug-compile: true`.

LioRei, 2022-08-21:
https://github.com/EscapeAutomate/Arduino.SignalR.Client/runs/7937885927?check_suite_focus=true

ArminJo, 2022-08-21:
Please try again, I updated the v3 version of the action.

LioRei, 2022-08-21:
it's working, thx :)

closed 2022-08-21 by LioRei