ArminJo/arduino-test-compile #32: [feature request] : accept "latest" as platform-version in ARDUINO_PLATFORM

tobozo, 2022-10-23:
Hi, 

thanks for this great project :+1: 

Some context: one of my workflows creates a multidimensional matrix from different combinations of platform/architectures/board/version.

I'm running arduino-test-compile a a step in that matrix:

```yml
    steps:
      - name: Checkout
        uses: actions/checkout@v2
        with:
          ref: ${{ github.event.pull_request.head.sha }}

      - name: Compile example
        uses: ArminJo/arduino-test-compile@v3
        with:
          arduino-board-fqbn: ${{matrix.platform}}:${{matrix.archi}}:${{matrix.board}}
          arduino-platform: ${{matrix.platform}}:${{matrix.archi}}@${{matrix.platform-version}}
          platform-url: ${{ matrix.platform-url }}
          required-libraries: ${{ matrix.required-libraries }}
          extra-arduino-cli-args: ${{ matrix.cli-args }}
```

The `${{matrix.***}}` variables are dynamic but the separators `:` and `@` aren't.
So if one of them is missing, the FQBN becomes invalid.

```yml
          arduino-platform: ${{matrix.platform}}:${{matrix.archi}}@${{matrix.platform-version}}
```

So far so good, everything works fine as long as I'm implicitely listing all package **versions** I want to be tested.

However this requires maintenance every time a new package version appears (e.g. esp32 recently released `2.0.5` package).

[image]

Here's what I tried to achieve when I was still full of illusions, believing the Arduino CI woud somehow read my mind :-)

### set `${{matrix.platform-version}}` to `latest`

[image]

[image]

Of course it fails! Why would `latest` be semver compliant? stupid me :facepalm: 

So here's my feature request: detecting a trailing `@latest` in the semver part of  `$ARDUINO_PLATFORM` and removing that trail from the FQBN so that Arduino CI just selects the latest version from the json package (which is the default behaviour) instead of erroring.

tobozo, 2022-10-23:
Here's the detached approach I'm using if it can be of any help, it blindly removes all instances of `@latest` from the concatenated `arduino-platform` string, I'm not sure how it behaves with arrays though.

```yml
- name: FBQN Check
  run: |
    export ARDUINO_PLATFORM="${${{matrix.platform}}:${{matrix.archi}}@${{matrix.platform-version}}/@latest/""}"
    echo "ARDUINO_PLATFORM=$ARDUINO_PLATFORM" >> $GITHUB_ENV

- name: Compile example
  uses: ArminJo/arduino-test-compile@v3
  with:
    arduino-board-fqbn: ${{matrix.platform}}:${{matrix.archi}}:${{matrix.board}}
    arduino-platform: ${{env.ARDUINO_PLATFORM}}
    platform-url: ${{ matrix.platform-url }}
    extra-arduino-cli-args: ${{ matrix.cli-args }}
    sketch-names: ${{ env.SKETCH_NAME }}
```

ArminJo, 2022-10-23:
Please try the updated version and provide a link to your 3 dimensional matrix build script, which I can include as a reference in my documentation.

closed 2022-10-23 by ArminJo

tobozo, 2022-10-23:
Thanks for the quick reply !

The workflow is still being edited, it's still focused on ESP32 boards so I'm not sure it would be a relevant example: 

https://github.com/tobozo/YAMLDuino/blob/1.2.9/.github/workflows/ArduinoBuild.yml

Some feedback after selecting `ArminJo/arduino-test-compile@master` in the workflow: 

Custom `@latest` version value in `arduino-platform` works as expected :partying_face: 

I'm also getting a lot of `Node.js 12 actions are deprecated` warnings.

[image]

I'm also occasionnaly getting some download errors, while this is probably more related to the network stability and submarines cutting undersea cables, I managed to capture one of those errors.

[image]

Of course the build fails consecutively, but the job reports another type of error which can be cryptic to understand the real cause is burried in the logs with more operations that had no reason to happen after that network failure.

It feels like the script could fail earlier e.g. right after ` Invalid argument passed: Found 0 platform for reference "esp32:esp32":` instead of going further.

[image]

Here's the [build run](https://github.com/tobozo/YAMLDuino/actions/runs/3307682666/jobs/5459492253) just in case you can access that.

ArminJo, 2022-10-23:
Thanks a lot for the suggestion of early exit 🥇 👍 
It is now implemented 😀

commit 867099ed (2022-10-23): The suffix @latest is always removed from specified arduino-platform. Fixes #32