ArminJo/arduino-test-compile #33: Using custom library fails to find the include file

BlueAndi, 2024-10-30:
This is the last github action log which failed:
https://github.com/BlueAndi/vscp-arduino/actions/runs/11603370091/job/32310195291#step:6:129

The repository contains a Arduino library, which shall be compiled together with the examples.
To avoid that the library is downloaded from the library manager, its included as custom library.

Unfortunately I run into the problem, that the vscp_platform.h header can not be found, see line 126. Additional the line seems not to be printed in utf-8.

ArminJo, 2024-10-31:
Is the vscp_platform.h in one of the directories  provided as parameters with "-I"?

per1234, 2024-10-31:
The problem is the incorrect path separator in the `#include` directive:

https://github.com/BlueAndi/vscp-arduino/blob/0a849affadce1ba88dcd251a2b8af4560c18ee0d/src/framework/events/vscp_evt_information.h#L51

```cpp
#include "..\user\vscp_platform.h"
```

The backslash is acting as the escape character instead of as a path separator. This is why you should always use the POSIX slash path separator in `#include` directives:

```cpp
#include "../user/vscp_platform.h"
```

This will work on all platforms (including Windows).

I see there are many occurrences of this bug in the library.

ArminJo, 2024-10-31:
@per1234 
Thank you very much for solving this issue 👍 .
And I am happy, that it was an issue of the user library, not the arduino-test-compile.

closed 2024-10-31 by ArminJo

BlueAndi, 2024-10-31:
@per1234 Thanks, I guess I had tomatos on the eyes. :-)
@ArminJo Yes, thats exactly why I would like to use arudino-test-compile. Thanks!

The main problem with the path in the includes was original that Arduino didn't support providing the include paths.